picoruby/prk_firmware #26: add modulo feature

kazhida, 2021-09-26:
There is an architecture called "modulo" that scans pins with I2C.
Although it is incomplete, I implemented the corresponding code including I2C.

branch:
https://github.com/kazhida/prk_firmware/tree/feature/modulo

compare:
https://github.com/picoruby/prk_firmware/compare/master...kazhida:feature/modulo?expand=1

hasumikin, 2021-09-26:
Thank you for the good suggestion 👍 

It'll take time for overall decision because as you already know, I2C is used in [pull/24](https://github.com/picoruby/prk_firmware/pull/24) and I think we should introduce a thing like `Kernel#require` of CRuby.
Please stay in touch with me to make a good code!

hasumikin, 2023-03-07:
@kazhida 
PRK Firmware 0.9.21 will be shipped with I2C class.
You can see an example here: https://github.com/picoruby/prk_firmware/wiki/Mouse#i2c
Please give it a try and reach out if you have questions and suggestions

(I will close this issue unless any comment is posted in a week)

closed 2023-03-14 by hasumikin