picoruby/prk_firmware #37: 0.9.5 may have a problem

geeknees, 2021-11-13:
Version 0.9.4 works fine, but 0.9.5 may have a problem. I'm using pipi_gherkin and the keymap is below.

https://github.com/geeknees/keymap/blob/main/prk_pipigherkin/keymap.rb

After upgrading, when I update the keymap, I get a situation where the keys of pause, f13, escape, and 2 are repeated. Here is a capture of the keystrokes for your reference. This is a recording of how the keyboard is connected to the PC and the corresponding key is input without typing.

[image]

I am reporting this, although it is not urgent, because the previous version works fine.
If there are any unclear points, please leave a comment so I can investigate.

hasumikin, 2021-11-15:
@geeknees 
Thanks for the report!
I couldn't reproduce exactly the same as what you reported though, there is a problem for sure.

This is the output on the serial console:
[image]

It will be an answer to your question of how to debug, you can see debug print from PicoRuby through a serial TTY (BAUD:115200 / Data:8bit / StopBits:1 / ParityCheck:none).

Anyway, #38 worked on my PiPi Gherkin.
Could you pull that branch, build it, and test whether your PiPi Gherkin works?
I will release 0.9.5.1 after your confirmation!

geeknees, 2021-11-15:
Thank you very much. I had a hard time building it in the mac environment, then I realized that there was a Dockerfile and solved it. Debugging was also confirmed. Thank you for your support.

*** 

build log
```
[  1%] Performing build step for 'ELF2UF2Build'
[100%] Built target elf2uf2
[  1%] No install step for 'ELF2UF2Build'
[  1%] Completed 'ELF2UF2Build'
[  4%] Built target ELF2UF2Build
[  4%] Performing build step for 'PioasmBuild'
[100%] Built target pioasm
[  5%] No install step for 'PioasmBuild'
[  6%] Completed 'PioasmBuild'
[  8%] Built target PioasmBuild
[  8%] Built target hal_posix

================================
building libpicorbc.a
  LIB_DIR=host-production/alloc_libc
lemon/lemon -p -DLEMON_PICORBC ./parse.y

102 parsing conflicts.
ld: library not found for -lcrt0.o
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[7]: *** [all] Error 1
make[6]: *** [compiler.o] Error 2
make[5]: *** [libpicorbc] Error 2
make[4]: *** [host-production/alloc_libc_libs] Error 2
make[3]: *** [src/mrubyc/src/mrblib.c] Error 2
make[2]: *** [lib/CMakeFiles/picorbc] Error 2
make[1]: *** [lib/CMakeFiles/picorbc.dir/all] Error 2
make: *** [all] Error 2
```

Dockerfile works brilliantly
` docker build -o keyboards --build-arg KEYBOARD=prk_pipigherkin`

serial TTY 
```
Autoreload is ready.
Suspending keymap.
Trying to reload keymap.
Initializing Keyboard ...
Overflow MAX_SYMBOLS_COUNT for 'offOverflow MAX_SYMBOLS_COUNT for 'entire_cols_size'
Overflow MAX_Overflow MAX_SYMBOLS_COUNT for 'entire_cols_size'
Overflow MAX_Overflow MAX_SYMBOLS_COUNT for 'entire_cols_size'
Overflow MAX_Overflow MAX_SYMBOLS_COUNT for 'entire_cols_size'
Overflow MAX_Overflow MAX_SYMBOLS_COUNT for 'entire_cols_size'
Overflow MAX_Overflow MAX_SYMBOLS_COUNT for 'a'
Overflow MAX_SYMBOLS_COUNT for 'c'
Starting keyboard task .Overflow MAX_SYMBOLS_COUNT for 'keycodes'
Undefined local variaOverflow MAX_SYMBOLS_COUNT for 'keycodes'
Undefined local variaOverflow MAX_SYMBOLS_COUNT for 'keycodes'
Undefined local variaOverflow MAX_SYMBOLS_COUNT for 'keycodes'
Undefined local variaOverflow MAX_SYMBOLS_COUNT for 'keycodes'
Undefined local variaOverflow MAX_SYMBOLS_COUNT for 'keycodes'
Undefined local variaOverflow MAX_SYMBOLS_COUNT for 'keycodes'
Undefined local variaOverflow MAX_SYMBOLS_COUNT for 'keycodes'
Undefined local variaOverflow MAX_SYMBOLS_COUNT for 'keycodes'
Undefined local variable or method 'clear' for FalseCOverflow MAX_SYMBOLS_COUNT for 'keycodes'
Undefined local variaOverflow MAX_SYMBOLS_COUNT for 'keycodes'
Undefined local variaOverflow MAX_SYMBOLS_COUNT for 'keycodes'
Undefined local variable or method 'clear' for FalseCOverflow MAX_SYMBOLS_COUNT for 'keycodes'
Undefined local variaOverflow MAX_SYMBOLS_COUNT for 'keycodes'
Undefined local variaOverflow MAX_SYMBOLS_COUNT for 'keycodes'
Undefined local variaOverflow MAX_SYMBOLS_COUNT for 'keycodes'
Undefined local variaOverflow MAX_SYMBOLS_COUNT for 'keycodes'
```

Thank you!

hasumikin, 2021-11-15:
Thank you very much!

PR #38 (2021-11-15): 🐛Hotfix: Increase MAX_SYMBOLS_COUNT
Fix #37
commit 99b95340 (2021-11-15): Fix #37
closed 2021-11-15 by hasumikin