picoruby/prk_firmware #43: Matrix to Layout Conversion

e3w2q, 2021-12-04:
Because the key matrix and the actual keyboard layout may be different, it would be nice to be able to write a keymap with a different layout from the key matrix, like the QMK Firmware layout macro.
https://beta.docs.qmk.fm/developing-qmk/c-development/hardware_keyboard_guidelines#less-than-keyboard_name.h-greater-than

If the Ruby syntax can handle this, I hope you can provide a sample of how to write it.

hasumikin, 2021-12-06:
@e3w2q 
Thanks for the suggestion.
I'm not 100% sure that I understood the concept of QMK's LAYOUT macro.
But I came to think something like this will be helpful:

```
kbd.add_layer_with :ANSI, :default, [
  nil,
  nil,
  %i[ KC_TAB ],
  nil,
  %i[ _ KC_AT ]
]
```

In this example,
- You can easily configure the ANSI layout for the default layer, and
  - The first, second, and fourth rows will be the same as the ANSI
  - `the first column of the third row` (which should be CapsLock by default) will be overridden with the `TAB` key
  - `the second column of the fifth row` (which should be Win/Apple by default) will be overridden with the `@` key
    - `_` in the fifth row just keeps the key on the boilerplate ANSI layout

Does this make sense?
Tell me if I'm off the point that you're talking about.

jpskenn, 2021-12-06:
When we made matrix, we can place keys (to PCB) anywhere we want.

Use 4 GPIO pins.
row pins A, B
col pins a, b

1. We can place keys in matrix order.
Aa  Ab
Ba  Bb

1. Also, we can place keys freely.(Different from matrix order)
Aa  Ba
Bb  Ab

In 1st case, we can write kaymap.rb as we look actual keyboard.

In 2nd case, can't imagine actual keyboard from keymap.rb.

e3w2q, 2021-12-06:
@hasumikin
Thanks for the reply.
My intentions are similar to @jpskenn .
For example, see the image in the middle of the following URL.
https://www.ibbotson.co.uk/hardware/keyboards/2020/12/06/firmware.html
This keyboard has a very different matrix and layout.
In QMK, in such a case, you can convert it with a layout macro, as shown in the following URL.
https://github.com/qmk/qmk_firmware/blob/master/keyboards/melody96/melody96.h
This allows you to set the keymap as per the layout, not as per the matrix, as shown in the following URL.
https://github.com/qmk/qmk_firmware/blob/master/keyboards/melody96/keymaps/default/keymap.c

hasumikin, 2021-12-07:
OK, now that I get it (hopefully)

So let's say,
Phisical layout:

```
Aa Ba
Bb Ab
```

Logical layout:

```
Aa Ab
Ba Bb
```

Then, you want to convert those through a conversion map something like this for example:

```ruby
kbd.add_layer_with_matrix :default,
                          %i[ KC_1 KC_2 KC_3 KC_4 ],
                          ConversionMap.new(
                            %i[ Aa Ba Bb Ab ],
                            %i[ Aa Ab Ba Bb ]
                          )
```

You can also write in this way:

```ruby
map = ConversionMap.new(%i[ Aa Ba Bb Ab ], %i[ Aa Ab Ba Bb ])
kbd.add_layer_with_conversion :default, %i[ KC_1     KC_2     KC_3     KC_4 ],    map
kbd.add_layer_with_conversion :raise,   %i[ KC_A     KC_B     KC_C     KC_D ],    map
kbd.add_layer_with_conversion :lower,   %i[ RGB_HUI, RGB_HUD, RGB_VAI, RGB_VAD ], map
```

How is this?

jpskenn, 2021-12-07:
Thanks.
That mapping looks nice.
I'll try it ASAP.

But, before trying, I need some learnig.
Because "keymap.rb" is my FIRST code with Ruby 😁

hasumikin, 2021-12-07:
That is just one idea.
Tell me if you think it will work enough even in a ​Duplex Matrix
and, if you come up with a better API.

Thanks!

jpskenn, 2021-12-07:
Sorry for my hasty decision.

I'll think about that.

jpskenn, 2021-12-07:
Is this mapping allows to remove unused matrix position?

Physical layout: *3 keys only*

```
Aa
Ba Bb
```
Logical layout:

```
Aa Ab
Ba Bb
```

Mapping:
```ruby
map = ConversionMap.new(
%i[
  Aa
  Ba Bb
],
%i[ 
  Aa XX
  Ba Bb
])
```

And we can write keymap as the keyboard looks.

```ruby
kbd.add_layer_with_conversion :default, %i[
  KC_1
  KC_2  KC_3
], map
```

Without mapping, assigning 'KC_NO' to keymap will result same.
```ruby
kbd.add_layer :default, %i[ 
  KC_1 KC_NO
  KC_2 KC_3
]
```
But the meaning of KC_NO should be "No keycode" or "Ignore(NOOP)" not "No key", I think.

hasumikin, 2021-12-08:
I don't think it makes sense.
I think the physical layout should have exactly four keys because there is a **possible point** that can be called `Ab` even if you didn't put a footprint of a key at the point.

In other words, *physical* shouldn't mean a layout of keys' footprint but a layout of the circuit.

jpskenn, 2021-12-08:
I understand your **possible point**.

With that, how do we define mapping 5*15 marix with unused 12 possible point? (= 60% keyboards)

Yes, Any order to fill unused point may work.
Is this non-strict rule acceptable?

hasumikin, 2021-12-08:
> how do we define mapping 5*15 marix with unused 12 possible point?

As you know, it is `KC_NO`.

This issue is to discuss converting logical layout to physical layout, right?
It should be really helpful when the difference between logical layout and physical one is quite big like your first example here https://github.com/picoruby/prk_firmware/issues/43#issuecomment-986763302
But skipping keys is not an essential topic, I guess.

jpskenn, 2021-12-08:
I

hasumikin, 2021-12-08:
I don't think it's a problem.
Please be back to the original issue.

jpskenn, 2021-12-08:
Ok. Let's go back.

Mapping idea at https://github.com/picoruby/prk_firmware/issues/43#issuecomment-987510193 is still looks nice.

My favorite is 2nd example

```ruby
map = ConversionMap.new(%i[ Aa Ba Bb Ab ], %i[ Aa Ab Ba Bb ])
kbd.add_layer_with_conversion :default, %i[ KC_1    KC_2    KC_3    KC_4 ],    map
kbd.add_layer_with_conversion :raise,   %i[ KC_A    KC_B    KC_C    KC_D ],    map
kbd.add_layer_with_conversion :lower,   %i[ RGB_HUI RGB_HUD RGB_VAI RGB_VAD ], map
```

The reason is ...

- Mapping and assigning keycodes are separated.
  It makes each lines simple.
- Easy to understand what is done by this.
- Logical layout and physical layout is paired.
  Not to forget another.

In addition to this, it is enough to work with Duplex Matrix.
With Duplex Matrix, `@rows.size` (or `@cols.size`) of new_map Array in add_layer_with_conversion method (function?) will be doubled, because Duplex Matrix scans its matrix bidirectionally.
This also affects to add_leyer method (function?).

This is my thoughts.
There's no problem exist now👍

hasumikin, 2021-12-08:
Thanks!
@e3w2q What do you think?

jpskenn, 2022-01-14:
This *more "physical"* implementation looks nice to me.
It may provide us freedom of pin assignment.

I have a minor concerning, not directory related to this new implement method.

- Matrix scan performance

  (I don't know well about hardware and internal logic of PRK's matrix scan...)

  How PRK scans each matrix point which defined at `kbd.init_matrix_pins`?
  Changing row at each matrix column?

  COL2ROW requires "wait delay" when change row.
  Too much change may affect matrix scanning performance.
  In QMK, it is 30ms.
  https://github.com/jpskenn/qmk_firmware/blob/e4d3ff2374142fe6431b68874f6604a471732fae/quantum/matrix_common.c#L8-L10
  https://github.com/jpskenn/qmk_firmware/blob/e4d3ff2374142fe6431b68874f6604a471732fae/quantum/matrix.c#L117-L121

hasumikin, 2022-01-15:
I'm not so worried.
The speed of RP2040 is always greater than I imagine 😀

closed 2022-01-22 by hasumikin