picoruby/prk_firmware #60: <<Request>> long-press operation

koyoko1127, 2022-02-11:
I want to activate different key codes by long-press and short-press

It is an operation that is not found in a normal PC keyboard, but it is a popular operation in embedded systems.
It can be used for careful operations and infrequent operations.

Thank you for the wonderful environment for me!
[image]

yswallow, 2022-02-11:
Although, it sends long-press keycode on UP after a long-press (not at the set time), 

```ruby
kbd.define_mode_key :SHORT_A_LONG_1, [ Proc.new { kbd.send_key(:KC_A) }, Proc.new { kbd.send_key(:KC_1) }, 500, nil ]
```

is not enough?

koyoko1127, 2022-02-11:
Wow　It's almost working!!

In my behavior, the keycode is entered with Long-press set time (500).  (Not key UP)
And, If the Long press time is set to 2000, key omissions will occur in continuous operations.
(short -> long -> short -> long ......)

Thanks [yswallow](https://github.com/yswallow)

hasumikin, 2022-02-13:
`Keyboard#define_mode_key` was basically designed to combine a normal key and modifier/layer hold key.
Maybe I should implement a new feature to fulfill @koyoko1127's request.
Let me think how to realize that because I'm personally excited that PRK Firmware will possibly be used in prototyping a real product 💪

koyoko1127, 2022-04-15:
Thank you for your aggressive suggestions.
If you are aiming for a multifunctional long press operation, you need to think about the context of use.
I have 4 stories（See figure below）
I think this includes Hi-dynamic range operations with repeat motion. Set the repeat interval time or set nil.

[image]
[image]

hasumikin, 2022-04-17:
@koyoko1127 
```
[a]      [A]      KC_NO      0
```

This is not a case this feature should cover because...

```
kbd.define_triple_function_key :BUTTON_A, :KC_A, 300, [:KC_LSFT, :KC_A], nil, nil
kbd.define_triple_function_key :BUTTON_B, :KC_B, 300, [:KC_LSFT, :KC_B], nil, nil
kbd.define_triple_function_key :BUTTON_C, :KC_C, 300, [:KC_LSFT, :KC_C], nil, nil
kbd.define_triple_function_key :BUTTON_D, :KC_D, 300, [:KC_LSFT, :KC_D], nil, nil
kbd.define_triple_function_key :BUTTON_E, :KC_E, 300, [:KC_LSFT, :KC_E], nil, nil
kbd.define_triple_function_key :BUTTON_F, :KC_F, 300, [:KC_LSFT, :KC_F], nil, nil
kbd.define_triple_function_key :BUTTON_G, :KC_G, 300, [:KC_LSFT, :KC_G], nil, nil
(...)
```

Nobody wants to write such a verbose code, right?
(Of course you can do it. But it's not supposed to do)

BTW, could you tell me more about "Hi-dynamic range operations"? I don't know it.

hasumikin, 2022-04-17:
Ah, but, you may be able to write like this, lol

```
("A".."Z").each do |c|
  kbd.define_triple_function_key "BUTTON_#{c}".to_sym, "KC_#{c}".to_sym, 300, [:KC_LSFT, "KC_#{c}".to_sym], nil, nil
end
```

Evel Ruby...

hasumikin, 2022-04-17:
Ah, but, you may be able to write like this, lol

```
("A".."Z").each do |c|
  kbd.define_triple_function_key "BUTTON_#{c}".to_sym, "KC_#{c}".to_sym, 300, [:KC_LSFT, "KC_#{c}".to_sym], nil, nil
end
```

Evil Ruby...

koyoko1127, 2022-04-17:
@hasumikin 

I don't know what " Hi-dynamic range operations " is officially called.

For example, when searching for an arbitrary value from 0 to 1000.
If you have a wide range of values ​​and want to set them in detail, you need both micro motion and coarse motion.

micro motion  (微動） ==> 0,1,2,3,4,....1000
coarse motion（粗動） ==> 0.20,40,60,80,.....1000   *It is better to have a repeat in the long-press

From now on, we need to control high resolution and large (long) size data in photo, CAD, and video.
Using multiple inputs with one key, I prototype an embedded product.

Thanks

hasumikin, 2022-04-18:
@koyoko1127 
Understood, thanks!
Let me think for a while