picoruby/prk_firmware #61: <<Request>> Add media keys (play / stop / next ...) - any workaround?

rudolf81, 2022-02-14:
Hi

Looks like media keys (play / stop / next ...) are not supported yet.

Please please please add them :D

Is there any workaround in the meantime to add media keys?

In the keymap.rb file, I tried to just add the decimal number for MediaPlayPause (179) instead of a named keycode, but it crashed.
https://docs.microsoft.com/en-us/dotnet/api/system.windows.forms.keys?view=windowsdesktop-6.0

`kbd.add_layer :default, %i[  KC_ESC    KC_1      KC_2      179      KC_4      ... ]`

Thank you very much!

hasumikin, 2022-02-15:
Thanks for the request.
Unfortunately, simply sending a media keycode won't work.
We need to implement a low-level HID feature so that media keys work.
I definitely have the plan to do that. Be patient!!!

rudolf81, 2022-02-15:
ah, thanks for the reply.

hasumikin, 2022-04-29:
| Symbol | Const in TinyUSB | Code |
|----|----|----|
| KC_MEDIA_PLAY_PAUSE | HID_USAGE_CONSUMER_PLAY_PAUSE       | 0X00CD |
| KC_MEDIA_NEXT_TRACK | HID_USAGE_CONSUMER_SCAN_NEXT        | 0X00B5 |
| KC_MEDIA_PREV_TRACK | HID_USAGE_CONSUMER_SCAN_PREVIOUS    | 0X00B6 |
| KC_MEDIA_STOP       | HID_USAGE_CONSUMER_STOP             | 0X00B7 |
| KC_AUDIO_VOL_UP     | HID_USAGE_CONSUMER_VOLUME_INCREMENT | 0X00E9 |
| KC_AUDIO_VOL_DOWN   | HID_USAGE_CONSUMER_VOLUME_DECREMENT | 0X00EA |

Symbols come from: https://github.com/qmk/qmk_firmware/blob/master/docs/keycodes.md
Consts come from: https://github.com/hathach/tinyusb/blob/master/src/class/hid/hid.h#L812-L822

hasumikin, 2022-08-01:
@rudolf81 
Coming 0.9.17 supports Consumer (Media) key
https://github.com/picoruby/prk_firmware/wiki/Consumer-key

PR #128 (2022-08-01): HID Consumer && more
Fix #61 
Fix #127 
Fix #114
closed 2022-08-01 by hasumikin