picoruby/prk_firmware #64: Support?

igreulich, 2022-02-27:
I could not find any official/proper support channels, so I am opening an issue here. If this is not the right place, I am sorry. Let me know where the right place is, and I will take my request there.

I have a hand-wired matrix with an inconsistent number of keys on each row, and column. The keyboard itself is/will be just a TKL.

Layout:
[image]

Matrix:
[image]

keycap.rb:
https://gist.github.com/igreulich/66b6f1168c312e3c95f6d82c332d86db

When I put the `keymap.rb` file on my Pico, the board does not work. I can connect to it via serial port, but it looks like it never fully loads. However the board does send input to the computer, at present it sends this input with out me pressing a key, and it is constant. (Currently the input seems to be `~` and `[TAB]`.

I did put `KC_NO` in place where there is no key in the matrix, but I don't know if the fact that there is no coordinates (from the perspective of the matrix) at those locations is breaking something?

I am concerned I have put the diodes on backwards, and that the fact that all of my rows don't have the same number of keys are my sticking points. However, I don't know enough to even know what all I don't know.

Thanks in advanced for any help!

hasumikin, 2022-02-28:
Your keymap.rb looks to work to me.
Although the `KC_NO` in {row:5, col: 12} might be wrong as you have an actual key switch at the position, it doesn't affect the whole functionality of the keyboard.
Could you show me the PCB data things like a schematic or even photos that express the implementation including diodes?

igreulich, 2022-03-02:
Let me know if I need more/clearer pictures; or anything else for that matter.

Thanks for all your help!

[image]
[image]
[image]

hasumikin, 2022-03-03:
Hand wiring!!!!

OK, your keyboard has a circuit so-called "row2col" while PRK basically supports "col2row".
The difference between col2row and row2col is the direction of the diodes.

PRK doesn't offer a short-handed way to configure the row2col matrix as of now though, no worries, you should be able to configure it by using `kbd.init_matrix_pins` instead of `kbd.init_pins`

The link below shows an example that runs a row2col keyboard.
https://github.com/swanmatch/prk_numatreus/blob/master/keymap.rb

You can see more info about `init_matrix_pins` here: https://github.com/picoruby/prk_firmware/wiki/Keyscan-matrix

igreulich, 2022-03-06:
> Hand wiring!!!!

I love it!

> OK, your keyboard has a circuit so-called "row2col" while PRK basically supports "col2row". The difference between col2row and row2col is the direction of the diodes.

Yeah, thats what I was figuring.

> PRK doesn't offer a short-handed way to configure the row2col matrix as of now though, no worries, you should be able to configure it by using `kbd.init_matrix_pins` instead of `kbd.init_pins`
 
> The link below shows an example that runs a row2col keyboard. https://github.com/swanmatch/prk_numatreus/blob/master/keymap.rb
 
> You can see more info about `init_matrix_pins` here: https://github.com/picoruby/prk_firmware/wiki/Keyscan-matrix

Thanks! Ill take a look, and report back!

igreulich, 2022-03-06:
That did the trick! [Here](https://gist.github.com/igreulich/268a7a23b8217cc391fe5bc8f63a2b41) is the final config, if you are interested!

Thanks for the help!

closed 2022-03-06 by igreulich