picoruby/prk_firmware #67: Split function and direct pins don't work together (version 0.9.10)

Elec-theo, 2022-04-01:
Hi,

For a split keyboard I want to use two RPi-Pico's and use the normal UART communication between them.
I'm testing with one empty Pico board.
With "kbd.set_anchor(:left)" or "(:right)" I define which side the Pico should be. When I use a matrix and init_pins the anchor works. But when I use direct pins (init_direct_pins) it always acts as the left side, even after setting the anchor to (:right).

I attached my testing code.

PS: I'm working with 0.9.10 since 0.9.11 doesn't support direct pins yet.

[keymap direct.rb.txt](https://github.com/picoruby/prk_firmware/files/8398507/keymap.direct.rb.txt)
[keymap matrix.rb.txt](https://github.com/picoruby/prk_firmware/files/8398508/keymap.matrix.rb.txt)

hasumikin, 2022-04-13:
@Elec-theo Thanks for the report.
Unfortunately, I myself don't have split-type-direct-pin-keyboard so can't fix the issue.
I leave this issue open until someone sends a patch or I could get such a keyboard...

Elec-theo, 2022-04-18:
> @Elec-theo Thanks for the report.
> Unfortunately, I myself don't have split-type-direct-pin-keyboard so can't fix the issue.
> I leave this issue open until someone sends a patch or I could get such a keyboard...

Thanks for looking into it. I am trying this with two bare RPi Pico’s soldered together with wires. If you have those maybe they can be used for debugging. I’d like to help, sadly I’m no programmer. 

Can I donate two RPi Pico’s to you to help you debug?

hasumikin, 2022-05-18:
@Elec-theo
A patch was merged. The coming release should work with split-type and direct-pin. Tell us the result on your end when you could check :)

PR #99 (2022-05-18): split on direct-pin works
fixed #67
closed 2022-05-18 by hasumikin