picoruby/prk_firmware #76: Dockerfile fails

pbrena, 2022-04-25:
Hello

Thanks for sharing your very interesting project!

I m interested in building a binary but, I guess the instructions in the wiki will only work in linux right? for me reading about it for the first time it was not obvious I tried and fail miserable in Macos catalina (suggestion add some pointers for the environment and requisites).

Then I tried and fail building the Dockerfile

last messages:

=> => transferring context: 2B                                                        0.0s
 => [build  2/16] RUN apt-get update && apt-get install --no-install-recommends -y   207.0s
 => [build  3/16] WORKDIR /prk_firmware                                                0.5s
 => [build  4/16] COPY . .                                                             0.3s
 => [build  5/16] WORKDIR /prk_firmware/src/ruby                                       0.4s
 => ERROR [build  6/16] RUN bundle install                                             1.2s
------
 > [build  6/16] RUN bundle install:
#10 1.161 Could not locate Gemfile
------
executor failed running [/bin/sh -c bundle install]: exit code: 10

hasumikin, 2022-04-27:
@pbrena Do you use an M1 Mac? I'm not sure but M1 Mac possibly doesn't work due to the Arm architecture

> (suggestion add some pointers for the environment and requisites).

OK.

hasumikin, 2022-04-27:
Ah, Catalina couldn't run on M1. So I totally don't know what happened on your end. Would you provide the full output message?

pbrena, 2022-04-29:
right x86 
[doker.txt](https://github.com/picoruby/prk_firmware/files/8592459/doker.txt)
[rakenativeonmacoos.txt](https://github.com/picoruby/prk_firmware/files/8592524/rakenativeonmacoos.txt)
thx

hasumikin, 2022-05-18:
@pbrena 
As far as we've tested, it could build successfully. The image below is the result on macOS Catalina:
[image]

Are you sure to follow this instruction?
https://github.com/picoruby/prk_firmware/blob/master/Dockerfile#L1-L11

pbrena, 2022-05-20:
i cleaned up and ran 
docker build . -t prk
but i still get

 => ERROR [6/6] RUN rake setup                                             0.7s
------
 > [6/6] RUN rake setup:
#11 0.604 rake aborted!
#11 0.604 No Rakefile found (looking for: rakefile, Rakefile, rakefile.rb, Rakefile.rb)
#11 0.604 
#11 0.604 (See full trace by running task with --trace)

so i cant Setup... i dont know what to do

hasumikin, 2022-05-20:
The message changed. It says you simply don't have Rakefile in the top directory.
I would think you have to learn programming basics, to be exact, how to use Linux/Unix command line and how to read an error message.
Really sorry but I am closing this issue because I believe the problem isn't on our end

closed 2022-05-20 by hasumikin

pbrena, 2022-05-20:
I understand and dont complain, Im a newbie and will keep on reading, and further learning.

Having said that, I just want to point out something...

As far as I understand the guide... the setup command (docker build . -t prk) is the very first to be run and the user has in principle no control of what is or isnt downloaded (particularly Rakefile in the top directory) without modifying the dokerfile itself and as the container isnt created and  no further command is posible... this should run as is, right?

I lost what my original deviation was so my approach was to cleanup my previos environment (following some online guides) to reinstall/upgrade Docker, (hence the change of message) and even re run the setup with --no-cache parameter to no avail...

I cannt identify the step in the dockerfile where top directory is populated but i will try to create an environment that gives a command prompt and proceed manually