picoruby/prk_firmware #78: <<Request>> Macro keys

SekoiaTree, 2022-04-27:
I would like to have a key where pressing it automatically writes \```, or some other list of characters. Composite keys seem close to what I want, but those press them at the same time instead of one after another. Is this possible?

hasumikin, 2022-04-27:
@SekoiaTree 
How about this?
```
kbd.define_mode_key :BACKTICKS, [ Proc.new { kbd.macro("```") }, nil, 200, nil]
```

SekoiaTree, 2022-04-27:
@hasumikin that... sort of works? First, it's a teensy bit slow (you can see the characters one by one), but second it's the wrong character. I think that has something to do with me being on a QWERTZ keyboard (this is for a macropad) instead of a QWERTY. I noticed that KC_Z outputs Y and inversely, for example.

SekoiaTree, 2022-04-27:
Also, for some reason it seems to press enter after? Like, I just tried `kbd.define_mode_key :BACKTICKS, [ Proc.new { kbd.macro("test") }, nil, 200, nil]`

and that outputs:
```
test

```

SekoiaTree, 2022-04-27:
So, I believe this line is the cause for the long time; maybe it should be configurable in the .macro call?
https://github.com/picoruby/prk_firmware/blob/eb4a068db8b1832ae7d5c16cf916ff458623d227/src/ruby/app/models/keyboard.rb#L1072

Second, I figured out the enter. If you add `,false` in the options, it turns it off. All good on that!

(entirely unrelated, but noticed when browsing the code. Why does it sleep for half a second when booting?
https://github.com/picoruby/prk_firmware/blob/cbb9502df336ff1642f90662700ce8b72f87d61b/src/ruby/app/models/keyboard.rb#L440
It also does it in a few other places)

hasumikin, 2022-04-28:
`Proc.new { kbd.macro("```") }` works for me and is faster than I type.
Whether it’s a QWERTY or not and whether it’s a full-size keyboard or a macro pad don’t matter at all

SekoiaTree, 2022-04-28:
Yes, yes it does. Any z inside the code instead types y, and inversely. Same for most special keys. This is an issue that pops up in a lot of things that don't handle non-qwerty keyboards, I think it's because of scancodes. And yes, it's faster than one can type, but it could be faster, so imo it should be configurable.

hasumikin, 2022-04-29:
Regarding the speed, I myself won’t make it configurable but will merge if you send a reasonable PR.

QWERZ, yeah I could get what is happening. Fixing the problem isn’t easy because of RAM consumption.
Furthermore, that problem can’t be a high priority because I don’t have any QWERZ keyboard. If many people came here to request fixing it, I will do 🫡

SekoiaTree, 2022-04-29:
Understandable, thanks! I'll try and do the configurable speed

hasumikin, 2022-04-30:
@SekoiaTree 
I came up with an idea of switching QWERTY to QWERTZ with adequate RAM consumption.
I've learned there are QWERTY, QWERTZ and AZERTY. Is there any other major kind of keyboard?
And also, I'm not knowledgeable about letters like `á` and `ü` etc.  Should a keyboard firmware consider them?

SekoiaTree, 2022-04-30:
Keyboard formats are really complex :/ there's like, one per language. A swiss qwertz keyboard isn't the same as a german qwertz keyboard for special keys, for example. Honestly, it might just be easier to have some custom config or utility that maps from what QWERTY has to whatever your keyboard is.

hasumikin, 2022-05-22:
I'm closing this because the original issue, which is Macro already been implemented.
Anyone can open another issue regarding suggesting QWERTZ

closed 2022-05-22 by hasumikin