picoruby/prk_firmware #86: Problem when pressing one mode_key while holding down another mode_key

alg0002, 2022-05-08:
There are key1 and key2.

 * key1 : mode_key for layer switching
 * key2 : mode_key in default layer, normal key after switching layers

The following is an example code.

```ruby
#                            key1      key2
kbd.add_layer :default, %i(SPC_LAY1, ENT_LCTL)
kbd.add_layer :layer1,  %i(XXXXXXXX, KC_BSPC )
kbd.define_mode_key :ENT_LCTL,[ :KC_ENTER, :KC_LCTL, 200, 200 ]
kbd.define_mode_key :SPC_LAY1,[ :KC_SPACE, :layer1,  200, 200 ]
```

In this case, I expected BackSpace to be typed by pressing key2 while holding down key1.
In reality, however, Enter is entered immediately after Ctrl+BackSpace is entered.
It seems that both the long-press key of the default layer and the key of layer1 are entered, and the short-press key of the default layer is entered immediately afterwards.

Note that if key2 is not mode_key but a normal key (e.g. KC_LCTL), the behavior is as expected.
This seems to happen only when the key before the layer switch is mode_key.

hasumikin, 2022-05-09:
There are two problems in your keymap.rb.

## 1. Syntax of Ruby  

```ruby
%i(SPC_LAY1, ENT_LCTL)
```

You should NOT write `,` in `%i()` expression. `%i(A B)` is a kind of syntax sugar for `[:A, :B]` 
See https://docs.ruby-lang.org/ja/3.0/doc/spec=2fliteral.html#percent

## 2. You misunderstood `:XXXXXXX` keycode

Your keymap goes back and forth between :default and :layer1 rapidly when you hold the :SPC_LAY1 key.
The effect of holding :SPC_LAY1 immediately disappears right after the layer changes to :layer1 because the key you are currently holding is :XXXXXXX which does NOT produce any side effect, there is no held layer, right?

You should've written as follow:

```ruby
kbd.add_layer :layer1,  %i(SPC_LAY1 KC_BSPC )
```

(Of course you have to delete `,` in `%i()`, too)

Eventually, your code should look like this:

```
kbd.add_layer :default, %i(SPC_LAY1 ENT_LCTL)
kbd.add_layer :layer1,  %i(SPC_LAY1 KC_BSPC )
```

## Besides, there is one more problem you will probably have

"Tapping the key1 while holding key2" doesn't work as you expect.
It is a specification matter of PRK.
PRK is now running on "hold-prioritized" logic.
Hard to explain what happens on it though, try it on your real keyboard.

Hope this will help you.

(Closing this issue but don't hesitate to reopen if you still have trouble)

closed 2022-05-09 by hasumikin

hasumikin, 2022-05-09:
This was already solved in this chat anyway... 🤔 
https://twitter.com/alg0002/status/1521401710954315776

alg0002, 2022-05-09:
I have already resolved this problem on Twitter, but I opened this issue because I thought it would be better to keep the information on Github as well.
Thank you for your response.