picoruby/prk_firmware #100: Input is looped.

marby3, 2022-05-19:
I am not familiar with the PRK Firmware specifications, but the input is currently looping.

code:
```
kbd.add_layer :default, %i(
  L_adjst  KC_1     KC_2     KC_3     KC_4     KC_5     KC_6     KC_7     KC_8     KC_9     KC_0     KC_MINUS KC_EQUAL KC_BSPC
  KC_TAB   KC_Q     KC_W     KC_E     KC_R     KC_T     KC_Y     KC_U     KC_I     KC_O     KC_P     KC_LBRC  KC_RBRC  KC_BSLS
  KC_LCTL  KC_A     KC_S     KC_D     KC_F     KC_G     KC_H     KC_J     KC_K     KC_L     KC_SCOLON KC_QUOT KC_ENT   XXXXXXX
  KC_LSFT  KC_Z     KC_X     KC_C     KC_V     KC_B     KC_N     KC_M     KC_COMM  KC_DOT   KC_SLSH  KC_RSFT  KC_UP    XXXXXXX
  XXXXXXX  XXXXXXX  KC_LGUI  KC_LALT  L_raise  KC_SPC   XXXXXXX  L_lower  XXXXXXX  KC_RALT  KC_RGUI  KC_LEFT  KC_DOWN  KC_RGHT
)
kbd.add_layer :raise, %i(
  XXXXXXX  KC_F1    KC_F2    KC_F3    KC_F4    KC_F5    KC_F6    KC_F7    KC_F8    KC_F9    KC_F10   KC_F11   KC_F12   KC_DEL
  KC_TAB   XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX
  KC_LCTL  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX
  KC_LSFT  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX  XXXXXXX  KC_RSFT  KC_HOME  XXXXXXX
  XXXXXXX  XXXXXXX  KC_LGUI  KC_LALT  XXXXXXX  KC_SPC   XXXXXXX  KC_SPC   XXXXXXX  KC_RALT  KC_RGUI  KC_PGUP  KC_END   KC_PGDN
)

kbd.define_mode_key :L_raise, [:KC_LANG1,  :raise,     200,    200 ]
kbd.define_mode_key :L_lower, [:KC_LANG2,  :lower,     200,    200 ]
```

I will use my code as a reference.
When I press L_lower while holding down the L_raise key, I can type KC_SPC.

At this time I can actually type Space, but when I release L_lower, Space continues to be typed.

Is this a spec behavior?

hasumikin, 2022-05-20:
Your keymap.rb looks to have some problems.
I think #86 gives you a clue. Read it first then ask us again if you still have trouble.

hasumikin, 2022-05-22:
Duplicate of #86

closed 2022-05-22 by hasumikin