picoruby/prk_firmware #101: Cannot build on M1 Mac

Taro-Hayashi, 2022-05-23:
I get the following error.
https://gist.github.com/Taro-Hayashi/cb3bf5fcbdd13c5e02cc76c89f3c2959

According to the error in Docker, the lemon (The Lemon Parser Generator)  executable is not supported?
I need someone else to try, as likely to be my shallow understanding or environmentally dependent. I have confirmed  Hello World in Ruby and Blink in PICO SDK.

hasumikin, 2022-05-23:
Thanks for the report!
Let me think more...

hasumikin, 2022-05-23:
@Taro-Hayashi Let me confirm, regarding "error without docker", did you follow the instruction "9.1.1. Installing the Toolchain" of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf#page37

hasumikin, 2022-05-23:
Regarding "WITH docker", try removing `prk_firmware/lib/picoruby/mrbgems/mruby-pico-compiler/lib/lemon` and then re-run command `prk_firmware % docker build --platform amd64 . -t prk`

Taro-Hayashi, 2022-05-23:
Cmake and arm-none-eabi-gcc are installed and up-to-date but I noticed VS CODE was not used. I’m sorry ignoring the procedure. 

And built successful with docker after removing lemon but could not run because it said there was no local image. 

Having probrem with tool usage,  I’ll try these after tomorrow and report back.
Thank you.

hasumikin, 2022-05-24:
@Taro-Hayashi 
I've fixed some points so that "WITHOUT docker" works.
Try setup again when you have time!

```
git pull origin master
rake deep_clean
rake setup
```

Taro-Hayashi, 2022-05-25:
Thanks for the update. I couldn't understand VS code and Docker finally, but in the terminal I was able to build and upgrade the firmware! 
I’d have added this to the wiki, but already it had been done. Thank you for everything.

By the way, I cloned the Meishi2 keymap in keyboards and tried to build with the keymap, I got error.
~~~
hayashi@Mac prk_firmware % rake build_with_keymap[prk_meishi2]
zsh: no matches found: build_with_keymap[prk_meishi2]
~~~
It may be I am wrong again, so feel free to close here.
Thank you.

Taro-Hayashi, 2022-05-25:
Duckduckgo for answer. I was in a hurry.
~~~
setopt nonomatch
~~~

hasumikin, 2022-05-28:
@Taro-Hayashi 
I made a possible solution of docker on M1 Mac.
Could you please try it when you have time?

```
rake deep_clean
git fetch origin fix_docker
git checkout fix_docker
docker compose build
docker compose run --rm prk rake
```

Taro-Hayashi, 2022-05-28:
I was told I need rake setup.
Did but got error. Please refer to.
https://gist.github.com/Taro-Hayashi/06aed2814b8c81b2f0be05de790a58e3

hasumikin, 2022-05-30:
@Taro-Hayashi 
I'm closing this issue because the building problem on the M1 Mac host was resolved.
It seems I myself have to have an M1 Mac and debug regarding docker.
Thanks anyway!

PR #103 (2022-05-30): Fix Docker
fix #101
closed 2022-05-30 by hasumikin