picoruby/prk_firmware #109: Roadmap for Mutual UART?

rschenk, 2022-06-19:
Hello! I am very interested to design a 3x5 split keyboard for PRK using Seeed XIAO RP2040 boards. I'm also a big fan of LEDs, and therefore would like to have mutual communication between the two halves.

I see that PRK currently has _experimental_ support for mutual UART over a single wire, and have read through the code and seen the warnings that it works but is potentially dangerous. Do you plan to continue to develop mutual communication over a single wire? Or will you use two wires in the future? Perhaps it is not known right now?

The reason I ask is, the XIAO RP2040 has exactly enough pins for a normal matrix, one led pin, and one uart pin. If two pins are needed for mutual uart in the future, then I would need to use a duplex matrix for the switches to have enough pins. So this is a design decision that I will need to make fairly early.

Thank you for this project! I'm very excited to use Ruby for this project!

hasumikin, 2022-06-20:
Thanks for the good question :)

I myself am using the Mutual UART for these seven months after releasing that feature, I've never experienced any damage to the RP2040.

Then, I think,

- Current single-wire mutual UART is going to be an official feature
- I don't have a plan to implement a two-wires UART
- Neither is I2C which uses two wires going to be supported (because nobody talked to me that somebody wants the feature)

So you can design your keyboard that fits current PRK APIs in terms of hardware.

When I release version 1.0.0 of PRK,

- `Keyboard#mutual_uart=` is going to be added as an alias of `Keyboard#mutual_uart_at_my_own_risk=`
- They are exactly the same thing
- Warning will also disappear

rschenk, 2022-06-20:
Fantastic! Thanks! I’ll keep you updated when I finish the board, and I hope to contribute to PRK as well. Cheers!

closed 2022-06-20 by rschenk

hasumikin, 2022-12-18:
@rschenk 
I changed my plan.
- From 0.9.20 (the next release),
- The original hardware one-way UART is removed
- Software mutual UART becomes the default and only option
- `Keyboard#mutual_uart_at_my_own_risk=` method works for backward compatibility but it no longer takes any effect

I decided as above regarding performance and source maintainability