picoruby/prk_firmware #114: composite key on encoder

methylpentane, 2022-07-10:
I want switch windows(tabs) by encoder but this code snippet won't work:

`
kbd.define_composite_key :NEXTTAB, %i(KC_LCTL KC_TAB)
kbd.define_composite_key :PREVTAB, %i(KC_LCTL KC_LSFT KC_TAB)
kbd.define_composite_key :NEXTWIN, %i(KC_LALT KC_TAB)
kbd.define_composite_key :PREVWIN, %i(KC_LALT KC_LSFT KC_TAB)

encoder_1 = RotaryEncoder.new(28, 27)
encoder_1.clockwise do
  kbd.send_key :NEXTWIN
end
encoder_1.counterclockwise do
  kbd.send_key :PREVWIN
end
kbd.append encoder_1

encoder_2 = RotaryEncoder.new(26, 22)
encoder_2.clockwise do
  kbd.send_key :NEXTTAB
end
encoder_2.counterclockwise do
  kbd.send_key :PREVTAB
end
kbd.append encoder_2
`

is it not supported, or any mistakes?

thanks

methylpentane, 2022-07-10:
I read keyboard.rb and understood that it's not supported sry
but I'll leave it open for a suggestion

hasumikin, 2022-07-31:
@methylpentane 
See `encoder_3` of this example: https://github.com/picoruby/prk_firmware/wiki/Rotary-encoder

This is going to be available from version 0.9.17

commit 1f82e37d (2022-07-31): To fix #114
PR #128 (2022-08-01): HID Consumer && more
Fix #61 
Fix #127 
Fix #114
closed 2022-08-01 by hasumikin