picoruby/prk_firmware #116: Problem with keymap not being recognized properly when connecting to Remap.

techmech-keeb, 2022-07-17:
I used PRKFirmware 0.9.14 and created the Firmware using the following document.
https://github.com/picoruby/prk_firmware/wiki/VIA-and-Remap

If I fill in the keymap directly in via_map.txt, it works fine.
On the other hand, when I connect to Remap, the keyboard is detected by the information in prk-conf.txt, but
the following anomaly occurs and I cannot change the keymap properly.

- The keymap displayed in Remap is different from what I set.
- When I edit the keymap in Remap and flash it, the changes are reflected in only some keys.
- When I flash a keymap in Remap, the file is corrupted.
- The key tester in Remap is not working.

#Windows10
#Chrome (103.0.5060.114)
#PRK Firmware (0.9.14)
#keyboard:OLSK60 (Adafruit KB2040)
https://github.com/techmech-keeb/OLSK60/tree/main/PRK_Firmware

yswallow, 2022-07-17:
Thank you for uploading your files.

You define the keyboard as 10 rows and 6 cols (as split keyboard, 5 rows and 12 cols) in `OLSK60_via.json`. But you define the keyboard as 5 rows and 13 cols in `keymap.rb`. 
And currently VIA on PRK is not supporting `nil` in key matrix. 
These cause first 3 problems.

And the key tester, is not keymap setting feature, is not implemented on PRK VIA.

techmech-keeb, 2022-07-17:
Thanks for the reply.

I have corrected the points noted below, but the behavior in Remap is not improved.
Is there anything else to keep in mind?

⇒Rewrote keymatrix in keymap.rb to 10 columns and 6 rows.
⇒Removed nil from keymap.rb keymatrix.
[image]

yswallow, 2022-07-18:
Ah. Sorry. VIA with `kbd.init_matrix_pins` has bug on `0.9.14`. This is fixed on `master`, but not released. I’ll build UF2 for you (in an hour). Please try it.

techmech-keeb, 2022-07-18:
Thank you very much. It is helpful.

Please let me know as well.
I have modified keymap.rb to support Remap this time.
However, when I edit the keymap directly, the physical location information is lost from keymap.rb, so I feel that the readability and workability is not so good.
If there is a better and smarter way to describe it, could you please let me know?

yswallow, 2022-07-18:
I built UF2, But the assigned keys fly away. It is a really bug.

[prk_firmware-0.9.14-20220718-91d43a2.uf2.zip](https://github.com/picoruby/prk_firmware/files/9128956/prk_firmware-0.9.14-20220718-91d43a2.uf2.zip)
.

yswallow, 2022-07-18:
> physical location information and Remap support

I have an idea, although it is not smart and not tested.

```ruby
kbd.init_matrix_pins(
  [
    [ [r0, c0], [r0, c1], [r0, c2], [r0, c3], [r0, c4], [r0, c5], [r5, c5], [r5, c4], [r5, c3], [r5, c2], [r5, c1],  [r5, c0] ],
    [ [r1, c0], [r1, c1], [r1, c2], [r1, c3], [r1, c4], [r1, c5], [r6, c5], [r6, c4], [r6, c3], [r6, c2], [r6, c1],  [r6, c0]],[[r9, c0],
      [r2, c0], [r2, c1], [r2, c2], [r2, c3], [r2, c4], [r2, c5], [r7, c5], [r7, c4], [r7, c3], [r7, c2], [r7, c1]],[[r7, c0]
      [r3, c0], [r3, c1], [r3, c2], [r3, c3], [r3, c4], [r3, c5], [r8, c5], [r8, c4], [r8, c3], [r8, c2], [r8, c1]],[[r8, c0],  [r9, c1],
      [r4, c0], [r4, c1], [r4, c2],           [r4, c3], [r4, c4],           [r4, c5],           [r9, c5], [r9, c4],  [r9, c3],  [r9, c2] ],
  ]
)
```

yswallow, 2022-07-18:
I modified `OLSK60_via.json`, `keymap.rb`, and `via_map.txt`. Please try it.

[OLSK60.zip](https://github.com/picoruby/prk_firmware/files/9130432/OLSK60.zip)

closed 2022-07-18 by techmech-keeb