picoruby/prk_firmware #122: The LED flickers momentarily and discontinuously only during the breath effect.

techmech-keeb, 2022-07-26:
The LED flickers momentarily and discontinuously only during the breath effect.
It does not depend on the value of Hue or Speed, and may occur when the Value is changed, but the reproducibility is not good.
This phenomenon does not occur with other effects.

https://user-images.githubusercontent.com/88352328/181024985-96310a65-fb6c-46dd-9b2e-84b3a0a4f976.MOV

hasumikin, 2022-07-27:
I thought this https://github.com/picoruby/prk_firmware/pull/85 has solved that issue 🤔 

@yswallow Could you tell me what you think?

yswallow, 2022-07-27:
@techmech-keeb Please tell us the blinking LED number (count from RP2040 GPIO) and LED product name (SK6812 or WS2812).

techmech-keeb, 2022-07-27:
Only one LED, SK6812mini-e, is connected.
(Although the circuit is designed to allow 22 pieces to be mounted, only one is currently mounted.)

yswallow, 2022-07-27:
hmm..  I can't reproduce this issue.

However, I found a bug on `ws2812.pio` and I try to fix. Can you try this UF2 ?
(I increased this `[1]` to `[3]` along with the SK6812 and WS2812 spec)
https://github.com/picoruby/prk_firmware/blob/48312d6b68240f8dd13c9f4aae1585c5b8b8e496/src/ws2812.pio#L9

[prk_firmware-0.9.16-20220727-6a47904.uf2.zip](https://github.com/picoruby/prk_firmware/files/9195443/prk_firmware-0.9.16-20220727-6a47904.uf2.zip)

hasumikin, 2022-07-27:
(I also couldn't reproduce)

techmech-keeb, 2022-07-27:
Thanks for your quick consideration.
I tried the version you provided.

The glitches were not completely corrected, but changed a little.

1. It occurs after manipulating RGB_VAI or RGB_VAD. It does not occur with other parameter changes.
2. It only occurs with breath effects.
3. In this version, after RGB_VAD input, it may flicker only once at a specific timing, or it may flicker periodically as before.
4. It may be my imagination, but it seems that the frequency of occurrence has slightly decreased.

yswallow, 2022-07-27:
How about this? (but I found first 8 leds color changing in delay, continue debugging...)
[prk_firmware-0.9.16-20220727-6a47904.uf2.zip](https://github.com/picoruby/prk_firmware/files/9196408/prk_firmware-0.9.16-20220727-6a47904.uf2.zip)

techmech-keeb, 2022-07-27:
Unfortunately, the situation did not change...

yswallow, 2022-07-27:
OK. I reproduced it. This happens whenever VAL is 15. Probably calculation bug. 

https://user-images.githubusercontent.com/1426202/181224545-ae2a7cf0-9a49-43a3-b3a3-cb1db4df9131.MOV

yswallow, 2022-07-27:
I think fixed. I'll put a pull-req.
[prk_firmware-0.9.16-20220727-6a47904.uf2.zip](https://github.com/picoruby/prk_firmware/files/9197542/prk_firmware-0.9.16-20220727-6a47904.uf2.zip)

techmech-keeb, 2022-07-27:
Amazing!!!
I have confirmed the problem by setting Val=15 in my environment.

And updating to the latest build eliminated the problem!

I will implement 22 LEDs later and check the operation again.
At least the problem no longer occurs in the current situation with one LED.

Thanks!

PR #123 (2022-07-27): Fix neopixel
* Fix #122
* expand WS2812 T1L to 0.8sec
* Fix LED update timing
closed 2022-07-27 by hasumikin