picoruby/prk_firmware #124: Request: Debug print improvements

techmech-keeb, 2022-07-28:
Is it possible to improve the debug print on the "USB Serial Port" so that the version of PRK Firmware can be displayed?
(as long as it does not affect the firmware boot speed)

[image]

hasumikin, 2022-07-28:
@techmech-keeb 
How is this?

```
===============================================
PRK Firmware 0.9.16 (20220728 revision 8f59e0e)
===============================================
```

This banner shows only when startup, not when reloading a keymap.rb. Do you think it's enough?

[image]

techmech-keeb, 2022-07-28:
That's cool.
I agree about the display timing.

PR #125 (2022-07-28): PRK_DESCRIPTION on serial console
Fix #124
closed 2022-07-28 by hasumikin