picoruby/prk_firmware #158: Large Duplex Matrix causing rp2040 to crash

doesntfazer, 2022-12-21:
I've noticed that when I do a large duplex matrix it crashes the RP2040 and the only way to get it to come back is to nuke the flash and put PRK back on it. 

Here is the keymap.rb that I used for testing just in case I did something wrong

```ruby
kbd = Keyboard.new

row0 = 1
row1 = 2
row2 = 3
row3 = 4
row4 = 5
row5 = 6
col0 = 7
col1 = 8
col2 = 9
col3 = 10
col4 = 11
col5 = 12
col6 = 13
col7 = 14
col8 = 16

# Init duplex matrix
kbd.init_matrix_pins(
  [
    [ [row0, col0], [col0, row0], [row0, col1], [col1, row0], [row0, col2], [col2, row0], [row0, col3], [col3, row0], [row0, col4], [col4, row0], [row0, col5], [col5, row0], [row0, col6], [col6, row0], [row0, col7], [col7, row0], [row0, col8], [col8, row0] ],
    [ [row1, col0], [col0, row1], [row1, col1], [col1, row1], [row1, col2], [col2, row1], [row1, col3], [col3, row1], [row1, col4], [col4, row1], [row1, col5], [col5, row1], [row1, col6], [col6, row1], [row1, col7], [col7, row1], [row1, col8], [col8, row1] ],
    [ [row2, col0], [col0, row2], [row2, col1], [col1, row2], [row2, col2], [col2, row2], [row2, col3], [col3, row2], [row2, col4], [col4, row2], [row2, col5], [col5, row2], [row2, col6], [col6, row2], [row2, col7], [col7, row2], [row2, col8], [col8, row2] ],
    [ [row3, col0], [col0, row3], [row3, col1], [col1, row3], [row3, col2], [col2, row3], [row3, col3], [col3, row3], [row3, col4], [col4, row3], [row3, col5], [col5, row3], [row3, col6], [col6, row3], [row3, col7], [col7, row3], [row3, col8], [col8, row3] ],
    [ [row4, col0], [col0, row4], [row4, col1], [col1, row4], [row4, col2], [col2, row4], [row4, col3], [col3, row4], [row4, col4], [col4, row4], [row4, col5], [col5, row4], [row4, col6], [col6, row4], [row4, col7], [col7, row4], [row4, col8], [col8, row4] ],
    [ [row5, col0], [col0, row5], [row5, col1], [col1, row5], [row5, col2], [col2, row5], [row5, col3], [col3, row5], [row5, col4], [col4, row5], [row5, col5], [col5, row5], [row5, col6], [col6, row5], [row5, col7], [col7, row5], [row5, col8], [col8, row5] ]
  ]
)
kbd.add_layer :default, %i(
      KC_1          KC_2          KC_3          KC_4     KC_5 KC_6 KC_7 KC_8 KC_9 KC_0 KC_1 KC_2 KC_3 KC_4 KC_5 KC_6 KC_7 KC_8
      KC_1          KC_2          KC_3          KC_4     KC_5 KC_6 KC_7 KC_8 KC_9 KC_0 KC_1 KC_2 KC_3 KC_4 KC_5 KC_6 KC_7 KC_8
      KC_1          KC_2          KC_3          KC_4     KC_5 KC_6 KC_7 KC_8 KC_9 KC_0 KC_1 KC_2 KC_3 KC_4 KC_5 KC_6 KC_7 KC_8
      KC_1          KC_2          KC_3          KC_4     KC_5 KC_6 KC_7 KC_8 KC_9 KC_0 KC_1 KC_2 KC_3 KC_4 KC_5 KC_6 KC_7 KC_8
      KC_1          KC_2          KC_3          KC_4     KC_5 KC_6 KC_7 KC_8 KC_9 KC_0 KC_1 KC_2 KC_3 KC_4 KC_5 KC_6 KC_7 KC_8
      KC_1          KC_2          KC_3          KC_4     KC_5 KC_6 KC_7 KC_8 KC_9 KC_0 KC_1 KC_2 KC_3 KC_4 KC_5 KC_6 KC_7 KC_8
)

kbd.start!
```

hasumikin, 2022-12-22:
Thank you for reporting.
I can't reproduce the issue because I myself don't have such a big duplex matrix.
So there would be little hope for a fix if that issue indeed comes from the duplex matrix.
If it does not, I will find someday, I guess

yswallow, 2022-12-22:
I found 2 reasons:

1. PRK cannot read too long line (over about 60 characters)
2. this `kbd.add_layer` crashes PRK

working `keymap.rb` is below:

```ruby

kbd = Keyboard.new

row0 = 1
row1 = 2
row2 = 3
row3 = 4
row4 = 5
row5 = 6
col0 = 7
col1 = 8
col2 = 9
col3 = 10
col4 = 17
col5 = 18
col6 = 13
col7 = 14
col8 = 16

# Init duplex matrix
kbd.init_matrix_pins(
  [
    [ [row0, col0], [col0, row0], [row0, col1], [col1, row0],
		[row0, col2], [col2, row0], [row0, col3], [col3, row0],
		[row0, col4], [col4, row0], [row0, col5], [col5, row0],
		[row0, col6], [col6, row0], [row0, col7], [col7, row0],
		[row0, col8], [col8, row0] ],
    [ [row1, col0], [col0, row1], [row1, col1], [col1, row1],
		[row1, col2], [col2, row1], [row1, col3], [col3, row1],
		[row1, col4], [col4, row1], [row1, col5], [col5, row1],
		[row1, col6], [col6, row1], [row1, col7], [col7, row1],
		[row1, col8], [col8, row1] ],
    [ [row2, col0], [col0, row2], [row2, col1], [col1, row2],
		[row2, col2], [col2, row2], [row2, col3], [col3, row2],
		[row2, col4], [col4, row2], [row2, col5], [col5, row2],
		[row2, col6], [col6, row2], [row2, col7], [col7, row2],
		[row2, col8], [col8, row2] ],
    [ [row3, col0], [col0, row3], [row3, col1], [col1, row3],
		[row3, col2], [col2, row3], [row3, col3], [col3, row3],
		[row3, col4], [col4, row3], [row3, col5], [col5, row3],
		[row3, col6], [col6, row3], [row3, col7], [col7, row3],
		[row3, col8], [col8, row3] ],
    [ [row4, col0], [col0, row4], [row4, col1], [col1, row4],
		[row4, col2], [col2, row4], [row4, col3], [col3, row4],
		[row4, col4], [col4, row4], [row4, col5], [col5, row4],
		[row4, col6], [col6, row4], [row4, col7], [col7, row4],
		[row4, col8], [col8, row4] ],
    [ [row5, col0], [col0, row5], [row5, col1], [col1, row5],
		[row5, col2], [col2, row5], [row5, col3], [col3, row5],
		[row5, col4], [col4, row5], [row5, col5], [col5, row5],
		[row5, col6], [col6, row5], [row5, col7], [col7, row5],
		[row5, col8], [col8, row5] ]
  ]
)

kbd.add_layer :default, %i(
KC_1 KC_2 KC_3 KC_4 KC_5 KC_6
KC_7 KC_8 KC_9 KC_0 KC_1 KC_2
KC_3 KC_4 KC_5 KC_6 KC_7 KC_8
KC_1 KC_2 KC_3 KC_4 KC_5 KC_6
KC_7 KC_8 KC_9 KC_0 KC_1 KC_2
KC_3 KC_4 KC_5 KC_6 KC_7 KC_8
KC_1 KC_2 KC_3 KC_4 KC_5 KC_6
KC_7 KC_8 KC_9 KC_0 KC_1 KC_2
KC_3 KC_4 KC_5 KC_6 KC_7 KC_8
KC_1 KC_2 KC_3 KC_4 KC_5 KC_6
KC_7 KC_8 KC_9 KC_0 KC_1 KC_2
KC_3 KC_4 KC_5 KC_6 KC_7 KC_8
KC_1 KC_2 KC_3 KC_4 KC_5 KC_6
KC_7 KC_8 KC_9 KC_0 KC_1 KC_2
KC_3 KC_4 KC_5 KC_6 KC_7 KC_8
KC_1 KC_2 KC_3 KC_4 KC_5 KC_6
KC_7 KC_8 KC_9 KC_0 KC_1 KC_2
KC_3 KC_4 KC_5 KC_6 KC_7 KC_8
)

# debug LED on XIAO RP2040
gpio_init 11
gpio_set_dir 11, Keyboard::GPIO_OUT_HI

rgb = RGB.new(12,1,0,false)
rgb.effect = :rainbow_mood
kbd.append rgb

kbd.start!
```

PRK crashes when I put `TAB` characters before the lines in `kbd.add_layer`.

example:
```ruby
kbd.add_layer :default, %i(
	KC_1 KC_2 KC_3 KC_4 KC_5 KC_6
	KC_7 KC_8 KC_9 KC_0 KC_1 KC_2
	KC_3 KC_4 KC_5 KC_6 KC_7 KC_8
...
```

doesntfazer, 2022-12-22:
**@yswallow that looks like it may have done the job! I will confirm in a bit to make sure it's all working correctly. But it doesn't crash the rp2040 at least.**

doesntfazer, 2022-12-23:
> 1.  PRK cannot read too long line (over about 60 characters)

Is this a limitation of Pico Ruby or PRK?

hasumikin, 2022-12-24:
PicoRuby

hasumikin, 2023-02-09:
I will close this issue unless another comment post in a week

closed 2023-02-20 by hasumikin