picoruby/prk_firmware #159: Dockerfile fails to build its container

hidsh, 2023-01-06:
Hello PRK,

I am interested in this next generation keyboard firmware!

So I've tried the Dockerfile in 0.9.20, but it failed `docker build`. It seems to be a problem related to `rake setup` according log.

I would appreciate it if you could fix this error.

Thank you.

```
~/prk/prk_firmware ❯❯❯ docker -v                                                                   ✘ 127 master ✱ ◼
Docker version 20.10.21, build baeda1f
~/prk/prk_firmware ❯❯❯ docker build -t test:1.0 -f Dockerfile .                                          master ✱ ◼
[+] Building 182.2s (12/13)                                                                                          => [internal] load build definition from Dockerfile                                                           0.0s
 => => transferring dockerfile: 658B                                                                           0.0s
 => [internal] load .dockerignore                                                                              0.0s
 => => transferring context: 2B                                                                                0.0s
 => [internal] load metadata for http://docker.io/manjarolinux/base:latest                                            3.3s
 => [auth] manjarolinux/base:pull token for [http://registry-1.docker.io](https://t.co/oumdBKZToB)                                               0.0s
 => [1/8] FROM http://docker.io/manjarolinux/base@sha256:9b74335bdc8ee94f816578ca3af04214fe9d024acba886fa62a0923a34  17.3s
 => => resolve http://docker.io/manjarolinux/base@sha256:9b74335bdc8ee94f816578ca3af04214fe9d024acba886fa62a0923a341  0.0s
 => => sha256:41ecfa93b6e0ebbc1d8f539fd538fc6a0a6ddc4b9f6dca050e3dacb3b39d8d54 901B / 901B                     0.0s
 => => sha256:82943290e75e46e5d4095862a63e45d942f12c738f7c984ac52b2a37b957004a 260.06MB / 260.06MB            10.3s
 => => sha256:9b74335bdc8ee94f816578ca3af04214fe9d024acba886fa62a0923a341b758c 741B / 741B                     0.0s
 => => sha256:769c82043b30e14fbd917b385c41170f9ca8ee1fd0c9d96c3bebc61a21f752bc 529B / 529B                     0.0s
 => => extracting sha256:82943290e75e46e5d4095862a63e45d942f12c738f7c984ac52b2a37b957004a                      6.5s
 => [internal] load build context                                                                              0.5s
 => => transferring context: 27.25MB                                                                           0.5s
 => [2/8] RUN pacman -Sy --noconfirm   cmake   arm-none-eabi-gcc   arm-none-eabi-newlib   ca-certificates    125.9s
 => [3/8] RUN git clone https://github.com/raspberrypi/pico-sdk.git -b 1.4.0                                   4.0s  => [4/8] RUN cd /pico-sdk/lib && git submodule update --init ./                                              29.7s  => [5/8] WORKDIR /prk_firmware                                                                                0.0s  => [6/8] COPY . .                                                                                             0.3s  => ERROR [7/8] RUN rake setup                                                                                 1.4s ------                                                                                                               > [7/8] RUN rake setup:                                                                                            #12 1.420 rake aborted!                                                                                             #12 1.420 Don't know how to build task 'setup' (See the list of available tasks with `rake --tasks`)
#12 1.422 
#12 1.422 (See full trace by running task with --trace)
------
executor failed running [/bin/sh -c rake setup]: exit code: 1
⏰182.82s total
~/prk/prk_firmware ❯❯❯                                                                               ✘ 1 master ✱ ◼
```

hasumikin, 2023-01-11:
@hidsh I fixed some points. Please try again when you have time

hidsh, 2023-01-13:
@hasumikin Thank you for your help!
I've confirmed that the `docker build` works as I expected on latest master branch.
Thanks.

```
~/prk/prk_firmware ❯❯❯ docker build -t test:1.0 -f Dockerfile .          master
[+] Building 375.8s (14/14) FINISHED
 => [internal] load build definition from Dockerfile                       0.0s
 => => transferring dockerfile: 658B                                       0.0s
 => [internal] load .dockerignore                                          0.0s
 => => transferring context: 2B                                            0.0s
 => [internal] load metadata for docker.io/manjarolinux/base:latest        4.6s
 => [auth] manjarolinux/base:pull token for registry-1.docker.io           0.0s
 => [1/8] FROM docker.io/manjarolinux/base@sha256:df27e66e0e5ab863edfb11  18.8s
 => => resolve docker.io/manjarolinux/base@sha256:df27e66e0e5ab863edfb111  0.0s
 => => sha256:df27e66e0e5ab863edfb1111d0127fe144a69f679705feb 741B / 741B  0.0s
 => => sha256:6e0d8461fd86e7e5de31d46309bb35a7ffc792f01e7359e 529B / 529B  0.0s
 => => sha256:c91e1883fd1767151b605f8b0a722f1bb33f5634d5a8a22 901B / 901B  0.0s
 => => sha256:f5705861d2e2a80fbd7eade264a09609947efeb 260.06MB / 260.06MB  9.6s
 => => extracting sha256:f5705861d2e2a80fbd7eade264a09609947efeb4f1a8f94f  8.6s
 => [internal] load build context                                          0.1s
 => => transferring context: 3.99MB                                        0.1s
 => [2/8] RUN pacman -Sy --noconfirm   cmake   arm-none-eabi-gcc   arm-  139.0s
 => [3/8] RUN git clone https://github.com/raspberrypi/pico-sdk.git -b 1.  4.8s
 => [4/8] RUN cd /pico-sdk/lib && git submodule update --init ./          31.4s
 => [5/8] WORKDIR /prk_firmware                                            0.0s
 => [6/8] COPY . .                                                         0.1s
 => [7/8] RUN rake setup                                                 164.9s
 => [8/8] RUN git config --global --add safe.directory /prk_firmware       0.5s
 => exporting to image                                                    11.5s
 => => exporting layers                                                   11.5s
 => => writing image sha256:0e3d4a9fa27fa0c93f031320b10584e984d627ea3f1ba  0.0s
 => => naming to docker.io/library/test:1.0                                0.0s
⏰377.36s total
~/prk/prk_firmware ❯❯❯
```

closed 2023-01-13 by hidsh