picoruby/prk_firmware #168: Corrupt FAT on FreeBSD / Linux

mneumann, 2023-04-30:
I just copied the most recent UF2 of prk_firmware to my RP2040 (from Rasperry Pi Foundation). On FreeBSD, it comes up like this:

```
ugen0.4: <PRK Firmware developers Default VID/PID> at usbus0
umodem0 on uhub0
umodem0: <PRK CDC> on usbus0
umodem0: data interface 1, has no CM over data, has no break
umass0 on uhub0
umass0: <PRK MSC> on usbus0
umass0:  SCSI over Bulk-Only; quirks = 0x0000
umass0:1:0: Attached to scbus1
uhid0 on uhub0
uhid0: <PRK Firmware developers Default VID/PID, class 239/2, rev 2.00/1.00, addr 4> on usbus0
da0 at umass-sim0 bus 0 scbus1 target 0 lun 0
da0: <TinyUSB Mass Storage 1.0> Removable Direct Access SCSI-2 device
da0: Serial Number 0.9.20 (20221226 revision ac046
da0: 1.000MB/s transfers
da0: 0MB (128 4096 byte sectors)
da0: quirks=0x2<NO_6_BYTE>
uhid1 on uhub0
uhid1: <PRK Firmware developers Default VID/PID, class 239/2, rev 2.00/1.00, addr 4> on usbus0
```

I can `dd if=/dev/da0 of=/tmp/prk.img bs=4096 count=128`. But I cannot mount it via `mount -t msdos /dev/da0 /mnt`. Is FreeBSD expecting a partition table?

On Ubuntu Linux (22.04), the FAT file system comes up, but it shows corrupted file names.

* Is it likely that during the transfer of the UF2 to the RP2040, corruption did take place? `prk_firmware` is quite "large"... so maybe chances for corruption is higher.
* Or is it more likely that the FAT implementation of `prk_firmware` is just missing some functionality?

Connecting against the serial port works: `picocom /dev/cuaU0` shows `Please make keymap.rb in PRKFirmware drive` repeatedly.

mneumann, 2023-04-30:
```sh
> minfo -i /tmp/prk.img ::
```

shows:

```
device information:
===================
filename="/tmp/prk.img"
sectors per track: 1
heads: 1
cylinders: 271

media byte: f8

mformat command line: mformat -t 271 -h 1 -s 1 -S 5 -i "/tmp/prk.img" ::

bootsector information
======================
banner:"MSDOS5.0"
sector size: 4096 bytes
cluster size: 1 sectors
reserved (boot) sectors: 1
fats: 1
max available root directory slots: 65408
small size: 271 sectors
media descriptor byte: 0xf8
sectors per fat: 1
sectors per track: 1
heads: 1
hidden sectors: 0
big size: 0 sectors
physical drive id: 0x80
reserved=0x1
dos4=0x29
serial number: 00001234
disk label="PRKFirmware"
disk type="FAT12   "
```

But using any other tool from `mtools` tells me `Cannot initialize '::'`.

hasumikin, 2023-04-30:
Have you tried resetting the flash ROM by flash_nuke.uf2? 
see https://github.com/picoruby/prk_firmware/wiki/FAQ#q-prkfirmware-drive-is-no-longer-mounted-after-installing-the-newest-release-why

yswallow, 2023-05-01:
It may caused by disk information collision.

PRK_Firmware drive reports capable of 512KB as a MassStorage, but reports 1MB as a FAT12 storage.
Actual capability is 512KB.

And the FAT implementation of PRK Firmware will be swapped on next release, so this issue may be solved.

I recommend you to copy `keymap.rb` only instead of copying whole disk image.

mneumann, 2023-05-02:
> Have you tried resetting the flash ROM by flash_nuke.uf2? see https://github.com/picoruby/prk_firmware/wiki/FAQ#q-prkfirmware-drive-is-no-longer-mounted-after-installing-the-newest-release-why

Just tried to reset it using the method you described. After the reset, the FAT looks good again (it's empty :).

But when I reflash with `prk_firmware-0.9.20-20221226-ac04670.uf2` I get the same issue. So I suspect, it has to do with the prk FAT implementation :)

mneumann, 2023-05-02:
> PRK_Firmware drive reports capable of 512KB as a MassStorage, but reports 1MB as a FAT12 storage. Actual capability is 512KB.

This is how the drive looks after `flash_nuke.uf2`:

```
da0 at umass-sim0 bus 0 scbus1 target 0 lun 0
da0: <RPI RP2 3> Removable Direct Access SCSI-2 device
da0: Serial Number E0C9125B0D9B
da0: 1.000MB/s transfers
da0: 128MB (262144 512 byte sectors)
da0: quirks=0x2<NO_6_BYTE>
```

And this is after flashing with `prk.....uf2` firmware:

```
da0 at umass-sim0 bus 0 scbus1 target 0 lun 0
da0: <TinyUSB Mass Storage 1.0> Removable Direct Access SCSI-2 device
da0: Serial Number 0.9.20 (20221226 revision ac046
da0: 1.000MB/s transfers
da0: 0MB (128 4096 byte sectors)
da0: quirks=0x2<NO_6_BYTE>
```

`prk` reports as 4k sectors whereas default Rasperry Pi Pico reports as 512 byte sectors (that's the "old" default sector size). Could that be the issue? 

> 
> And the FAT implementation of PRK Firmware will be swapped on next release, so this issue may be solved.

Cool! Thanks!

> I recommend you to copy `keymap.rb` only instead of copying whole disk image.

Hmm, how would that work if I cannot flash a valid `prk` firmware?

yswallow, 2023-05-02:
Sorry, I misread `if` and `of` on `dd`.

PRK reports storage size too large (1MB) on FAT. This may make FreeBSD request outside the flash.

I think the `dd` is based on MassStorage (basic layer of the storage), so it requests only 512KB, it works.

> how would that work if I cannot flash a valid `prk` firmware?

OK, I'll try making valid FAT image. Please wait.

yswallow, 2023-05-02:
Please try this [FAT512K.img](https://github.com/picoruby/prk_firmware/files/11371568/FAT512K.zip) (zipped for GitHub) on FreeBSD.

ex: `dd if=FAT512K.img of=/dev/da0` (`da0` is prk drive)

mneumann, 2023-05-02:
Still fails.

After `dd if=FAT512K.img of=/dev/da0` (and unplugging USB and plugging it in again):

```
da0 at umass-sim0 bus 0 scbus1 target 0 lun 0
da0: <TinyUSB Mass Storage 1.0> Removable Direct Access SCSI-2 device
da0: Serial Number 0.9.20 (20221226 revision ac046
da0: 1.000MB/s transfers
da0: 0MB (128 4096 byte sectors)
da0: quirks=0x2<NO_6_BYTE>
```

And mounting it with `mount -t msdos /dev/da0 /mnt` fails:

```
mount_msdosfs: /dev/da0: Invalid argument
```

Are you sure, you are not messing with 512-bytes vs. 4096-bytes sector sizes?  Or is TinyUSB Mass Storage library always using 4096-bytes sized disk sectors?

yswallow, 2023-05-02:
Yes, I didn’t specify sector size, because I don’t know which value is the best. 

And, in my newly-installed FreeBSD 13.2 environment, `mount -t msdos /dev/da0 /mnt` success with no message. It’s a difficult problem….

mneumann, 2023-05-02:
Interesting! :)

Can you show me your `dmesg` output related to `da0`? I also have most recent FreeBSD 13.2 installed. Do you think this is a software issue or could it also be hardware-related? Is it possible to have data corruption when transferring UF2?

hasumikin, 2023-05-03:
> Or is TinyUSB Mass Storage library always using 4096-bytes sized disk sectors?

No, it is not TinyUSB's reason.
The flash ROM of RP2040 requires aligned 4096 bytes as a deletion unit. We only can delete some data by clearing 4096 bytes area at once. This is the reason for the 4096 sector size.
I'm now implementing an SD card FAT filesystem that has 512 bytes sector size and it can mount through TinyUSB

yswallow, 2023-05-03:
The dmesg was:

```
da0 at umass-sim0 bus 0 scbus4 target 0 lun 0
da0: <TinyUSB Mass Storage 1.0> Removable Direct Access SCSI-2 device
da0: Serial Number 0.9.20 (20221226 revision ac046
da0: 1.000MB/s transfers
da0: 0MB (128 4096 byte sectors)
da0: quirks=0x2<NO_6_BYTE>
```

And sorry, I have tested `dd` with no specified `bs` on Manjaro Linux, but `dd` with no specified `bs` on FreeBSD was wrong.

I tried `dd if=/dev/FAT512K.bin of=/dev/da0 bs=4k` with [new bin file here](https://github.com/picoruby/prk_firmware/files/11378856/FAT512K.zip) on FreeBSD, and I could make PRK working.
Please try it.

hasumikin, 2023-05-03:
@mneumann 
Could you try this uf2? It's a candidate for the next release
https://drive.google.com/file/d/1w8fy_2akx3asZxyj8zGdRNPcrkATFkyl/view?usp=share_link

mneumann, 2023-05-03:
> I tried `dd if=/dev/FAT512K.bin of=/dev/da0 bs=4k` with [new bin file here](https://github.com/picoruby/prk_firmware/files/11378856/FAT512K.zip) on FreeBSD, and I could make PRK working.
> Please try it.

This works! So using `bs=4k` is the trick here.

mneumann, 2023-05-03:
> Could you try this uf2? It's a candidate for the next release
> https://drive.google.com/file/d/1w8fy_2akx3asZxyj8zGdRNPcrkATFkyl/view?usp=share_link

This works! With the new firmware, I can mount `da0s1`. Thanks a lot!

hasumikin, 2023-05-08:
Closing as it looks solved

closed 2023-05-08 by hasumikin

mneumann, 2023-05-08:
Yes, solved! Thanks!